kitesurfer1404/WS2812FX #342: Led transitions problem on esp32

chakjer, 2023-11-07:
I use transition in my program and i found that after some times v2 effect just stop. 
In next v2 period it still show hold v2 previous effect.
It is really hard to find which effect causes it but few effect later and it backs to normal.
Main program run on core1, core0 is for the rest( www,mqtt,http ect).

moose4lord, 2023-11-08:
Can you cut/paste your code and describe the hardware you're using?

chakjer, 2023-11-12:
Hello,
I think i found a problem, every time settings for segment will be change segment need to be started.
I added ` ledy.v2->start();` after `ledy.v2->setSegment...` and up till now it helps, 3 days of testings and problem didn't occur.
Of course, I have those line in code after initialization.

moose4lord, 2023-11-13:
If you are going to dynamically change the segment settings (i.e. in the loop() function), then you should run the clear() and resetSegments() functions to reset the LED data. Do this **before** changing the segment settings, like so:
```
  ws2812fxt.v1->clear();          // reset LED pixel data
  ws2812fxt.v1->resetSegments();  // reset runtime data

  ws2812fxt.v1->setSegment(0, 0, LED_COUNT-1, newMode, GREEN, 5000); // change segment settings
```

closed 2023-11-14 by chakjer