kitesurfer1404/WS2812FX #344: 2 Neopixel Squares 8x8 change sequence (Solution of issue #284 not working)

Marineflieger, 2024-01-07:
Hello, so i've been trying all day to change the sequence of pixels and nothing 
works. Maybe I'm mixing up the terms, but I think the explanation makes it clear what my goal is.
The initial situation is that i have 2 neopixel squares with 8x8 pixels each that are horizontally next to each other (first left, second right; the 2nd square is connected to the output of the first; both have pixel 0 top left). When the Larson scanner is running, for example, it should start in the first row of the first square and continue in the first row of the second square. So basically double my "screen".
If I now reorder the pixels, as in issue #284:

(for testing I used the only the first row of the second square)
#define LED_COUNT 72

uint16_t ledMap[LED_COUNT] = {0,1,2,3,4,5,6,7,64,65,66,67,68,69,70,71, 8,9,10, [etc...]60,61,62,63};

then it skips the second row (remains dark), continues at the third and executes the second row at the end.
So, not exactly what I want. I can't figure out what I need to change either. In my opinion, it can't work this way, as this solution changes things on the other end as I need it.
There seem to be antoher solution with Adafruit_NeoMatrix for tiled matrices. I just don't know how to get the effects of this great library to work with it, Help!

moose4lord, 2024-01-08:
I think your ledMap[] array is not setup to do what you want. If you want to map the second square so that it starts after the first row of the first square, your ledMap[] array would be something like this:
```c++
uint16_t ledMap[LED_COUNT] = {
   0, 1, 2, 3, 4, 5, 6, 7,
  16,17,18,19,20,21,22,23,
  24,25,26,27,28,29,30,31,
  32,33,34,35,36,37,38,39,
  40,41,42,43,44,45,46,47,
  48,49,50,51,52,53,54,55,
  56,57,58,59,60,61,62,63,
  64,65,66,67,68,69,70,71,
   8, 9,10,11,12,13,14,15
};
```

Also, not related to your problem, but I notice a bug in the demo sketch from #284. The for loop that does the led mapping is executing too many times and should be changed from this:
```c++
  for (int i=0; i < ws2812fx_p.getNumBytes(); i++) {
    ws2812fx_p.setPixelColor(i, ws2812fx_v.getPixelColor(ledMap[i]));
  }
```
to this:
```c++
  for (int i=0; i < LED_COUNT; i++) {
    ws2812fx_p.setPixelColor(i, ws2812fx_v.getPixelColor(ledMap[i]));
  }
```

Marineflieger, 2024-01-08:
Uff, I couldn't see the wood for the trees -  this makes perfectly sense now. 
Thank you so much for helping me out.