kitesurfer1404/WS2812FX #348: question ws281212b

anto168, 2024-02-23:
hiii good evening, sorry to interrupt or maybe on a different path.  question, I have a problem with the ws2812b code, maybe here I can help me.  using the fastled library.  to run several leds with running then static turning them all on for 1 second then to return to the initial loop I had difficulty, it didn't return correctly.  here is my code.  it's easy to help....

`   #include "FastLED.h"
#define NUM_LEDS  60    // jumlah LED pada strip
#define PIN       2     // Pin LED

CRGB leds[NUM_LEDS];

unsigned long previousMillis = 0;           // Menyimpan waktu terakhir LED diperbarui
int count = 0;                              // Menyimpan hitungan untuk penambahan hingga NUM_LEDs
bool animationCompleted = false;            // Menyimpan status apakah siklus animasi sudah selesai

void setup() {
  FastLED.addLeds<WS2812B, PIN, GRB>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setMaxPowerInVoltsAndMilliamps(5, 1500);    //daya strip LED 5V, 1500mA
  FastLED.clear();                                    // semua LED menjadi "MATI"
}

void loop() {
  if (!animationCompleted) {
    // Efek animasi
    shootingStarAnimation(255, 255, 255, random(10, 60), random(5, 40), random(2000, 8000), 1);
  } else {
    // Jika animasi selesai, nyalakan semua LED
    turnOnAllLEDs();
    // Set variabel animationCompleted menjadi true agar program tidak masuk ke dalam blok ini lagi
    animationCompleted = true;
  }
}

void shootingStarAnimation(int red, int green, int blue, int tail_length, int delay_duration, int interval, int direction) {
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    count = 0;      // Reset interval
  }
  if (direction == -1) {        // Opsi arah mundur untuk LED
    if (count < NUM_LEDS) {
      leds[NUM_LEDS - (count % (NUM_LEDS+1))].setRGB(red, green, blue);    // Atur LED dengan nilai warna
      count++;
    }
  }
  else {
    if (count < NUM_LEDS) {     // Opsi arah maju untuk LED
      leds[count % (NUM_LEDS+1)].setRGB(red, green, blue);    // Atur LED dengan nilai warna
      count++;
    }
  }
  fadeToBlackBy(leds, NUM_LEDS, tail_length);                 // Memudar LED ekor menjadi hitam
  FastLED.show();
  delay(delay_duration);                                      // Jeda untuk menentukan kecepatan animasi
  // Jika animasi sudah mencapai akhir, tandai bahwa animasi sudah selesai
  if (count >= NUM_LEDS) {
    animationCompleted = true;
  }
}

void turnOnAllLEDs() {
  // Nyalakan semua LED dengan warna putih
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::White;
  }
  FastLED.show();
}
 `

moose4lord, 2024-02-23:
This isn't a FastLED library discussion, but I think I can point you in the right direction.
It looks like you're animationCompleted logic isn't correct. I think it makes sense to move some of that logic out of the shootingStarAnimation() function and put it in the loop() function. I came up with this:
```cpp
#include "FastLED.h"
#define NUM_LEDS 60    // jumlah LED pada strip
#define PIN 2          // Pin LED

CRGB leds[NUM_LEDS];

unsigned long previousMillis = 0;  // Menyimpan waktu terakhir LED diperbarui
int count = 0;                     // Menyimpan hitungan untuk penambahan hingga NUM_LEDs
//bool animationCompleted = false;   // Menyimpan status apakah siklus animasi sudah selesai

void setup() {
//FastLED.addLeds<WS2812B, PIN, GRB>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);  // creates a purple color cast??
  FastLED.addLeds<WS2812B, PIN, GRB>(leds, NUM_LEDS).setCorrection(UncorrectedColor);
  
//FastLED.setMaxPowerInVoltsAndMilliamps(5, 1500);  //daya strip LED 5V, 1500mA
  FastLED.setMaxPowerInVoltsAndMilliamps(5, 200);   //daya strip LED 5V, 200mA
  FastLED.setBrightness(8);                         // dim LEDs
  FastLED.clear();                                  // semua LED menjadi "MATI"
}

void loop() {
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis < 10000) {  // run star animation for 10 seconds
    shootingStarAnimation(255, 255, 255, random(10, 60), random(5, 40), random(2000, 8000), 1);
  } else {
    turnOnAllLEDs();
    previousMillis = currentMillis;  // reset timer
    count = 0;
  }
}

void shootingStarAnimation(int red, int green, int blue, int tail_length, int delay_duration, int interval, int direction) {
  count = (count + 1) % NUM_LEDS;

  if (direction == -1) {  // Opsi arah mundur untuk LED
      leds[NUM_LEDS - count - 1].setRGB(red, green, blue);  // Atur LED dengan nilai warna
  } else {
      leds[count].setRGB(red, green, blue);  // Atur LED dengan nilai warna
  }
  fadeToBlackBy(leds, NUM_LEDS, tail_length);  // Memudar LED ekor menjadi hitam
  FastLED.show();
  delay(delay_duration);  // Jeda untuk menentukan kecepatan animasi
}

void turnOnAllLEDs() {
  // Nyalakan semua LED dengan warna putih
  fill_solid(leds, NUM_LEDS, CRGB::White);
  FastLED.show();
  delay(1000);  // show static LEDs for 1 second
}
```

anto168, 2024-02-26:
Thank you for your help, with a few changes to the code you provided, I was able to find the animation form I wanted.  my problem is closed, thank you very much...