kitesurfer1404/WS2812FX #350: playing a bit

ldijkman, 2024-03-22:
playing a bit with auto mdns list

do not understand yet why there is no on off
and why select color background of input is not colored to selected

https://ldijkman.github.io/async-esp-fs-webserver/

[image]

moose4lord, 2024-03-22:
Are you trying to use the esp8266_interface example sketch with webSockets? If so, that won't work. The example sketch is expecting ws2812 data to come across the network as simple HTTP query parameters. Like this:
```c++
http://ws2812.local/set?c=1946112
```
A webSocket data frame is incompatible. You'd need to write some custom code to translate the webSocket data into parameters that the esp8266_interface code can understand.

ldijkman, 2024-03-22:
It is working
Ok if you click on rgb 000 it is off
Maybe a button for off
I was expecting color in rgb button

[image]

moose4lord, 2024-06-15:
Closing this issue as part of periodic cleanup.

closed 2024-06-15 by moose4lord