kitesurfer1404/WS2812FX #351: flash install from bowser on 4mb ESP8266 12E/12F gpio2=D4 20leds

ldijkman, 2024-03-23:
a test maybe ...

on
- https://ldijkman.github.io/async-esp-fs-webserver/
is a button
- Connect & Install on 4mb ESP2866 12E / 12F ws2812 led strip on gpio2 = D4 20 leds

test to Easy flash ESP from in browser

server contains /2812.html  or http://ws2812.local/2812.html
wich can set led color for now

20 leds gpio2 = D4

[image]

ldijkman, 2024-03-23:
video with a bit more chaotic info

https://youtu.be/YS3hWK9fHIE

ldijkman, 2024-03-24:
bit better video
- flash from browser
- https://youtu.be/5SzWE3oxlLI

ldijkman, 2024-03-25:
bit further 
got some buttons now

new video https://youtu.be/xio0c5hPp5I?si=v2go6OXg9qnJbuRZ

https://ldijkman.github.io/async-esp-fs-webserver/

[image]

ldijkman, 2024-03-26:
[image]

ldijkman, 2024-03-27:
added settings feedback
speed brightness mode

https://ldijkman.github.io/async-esp-fs-webserver/

[image]

ldijkman, 2024-03-27:
added a speed slider

https://ldijkman.github.io/async-esp-fs-webserver/

[image]

ldijkman, 2024-03-30:
some changes 
higlight button on auto play cycle

https://ldijkman.github.io/async-esp-fs-webserver/

[image]

moose4lord, 2024-06-15:
Closing this issue as part of periodic cleanup.

closed 2024-06-15 by moose4lord