kitesurfer1404/WS2812FX #354: 1.4.3 Update breaks ESP8266/ESP32 compiling with Arduino IDE

ESurge, 2024-06-16:
The 1.4.3 update breaks ESP32 and ESP8266 compiling when using Arduino IDE. [Error log attached](https://github.com/user-attachments/files/15857416/WS2812FX-1.4.3-ESP_Errors.txt)

How to reproduce:
- Open up Arduino IDE (I'm using 2.3.2)
  - WS2812FX 1.4.3 is installed
  - I don't think this matters but I'm using esp32 board package version 3.0.1 from Espressif Systems, and also esp8266 board package version 3.1.2 from ESP8266 Community
- Open WS2812FX example: `esp8266_webinterface`
- Set Board to ESP8266 or ESP32
- Compile

Workaround is to go back to 1.4.2 and compile.

Side fix: Reverting changes regarding (uint8_t) within some max() functions in modes.cpp, and modes_funcs.cpp allows compilation.

moose4lord, 2024-06-16:
Thanks for catching this. I should have done more testing with the ESP tool chain. Working on a fix that should be released soon.

moose4lord, 2024-06-16:
I've released v1.4.4 to fix this issue. Let me know if you find any more dumb mistakes. :)

ESurge, 2024-06-16:
This is fixed with 1.4.4

Thanks for the quick response!

closed 2024-06-16 by ESurge