kitesurfer1404/WS2812FX #356: ESP32 keep rebooting

KaTOxDev, 2024-06-25:
Hello , i use ESP32 with this example : 
`#include <WS2812FX.h>

#define LED_COUNT 120
#define LED_PIN 17

#define TIMER_MS 5000

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);

unsigned long last_change = 0;
unsigned long now = 0;

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(1000);
  ws2812fx.setColor(0x007BFF);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();
}

void loop() {
  now = millis();

  ws2812fx.service();

  if(now - last_change > TIMER_MS) {
    ws2812fx.setMode((ws2812fx.getMode() + 1) % ws2812fx.getModeCount());
    last_change = now;
  }
}
`

when i set LED_COUNT to 120 i get this error and my ESP32 keep rebooting 

`***ERROR*** A stack overflow in task loopTask has been detected.

Backtrace: 0x400827bd:0x3ffaf270 0x40087e15:0x3ffaf290 0x4008ac01:0x3ffaf2b0 0x400890f2:0x3ffaf330 0x4008adb8:0x3ffaf360 0x4008ad68:0x3ffaf380 0x40084336:0x00000001 |<-CORRUPTED

ELF file SHA256: 7a1aac9b16b5226f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c`

i try to set LED_COUNT  to 30 and it work with no error .
idk the limit of LED_COUNT

moose4lord, 2024-06-25:
This seems to be a bug caused by how the Adafruit-NeoPixel library is programming the ESP32's RMT hardware when arduino-esp32 core v3.0 is installed. I'll look into this further, but in the meantime I would suggest using Board Manager to downgrade the arduino-esp32 core to version 2.0.17. That should get you up and running.

moose4lord, 2024-06-25:
Looks like this issue is already being discussed on the Adafruit_Neopixel GitHub repository:
[Adafruit_NeoPixel/issues/375](https://github.com/adafruit/Adafruit_NeoPixel/issues/375)
There's already a proposed fix and pull request, so it should be fixed soon.

moose4lord, 2025-01-21:
Fixed in Adafruit_NeoPixel v1.12.4. Closing.

closed 2025-01-21 by moose4lord