kitesurfer1404/WS2812FX #358: custom effecrt

TheYigidOS, 2024-08-03:
hi how can i set color from this ["00c3cc","0e131f","3372e3","ff0035","7426ef"] nearl all modes just uses first 2 index of the array but i want to make it whole array thank u!

moose4lord, 2024-08-04:
You're right, the pre-built WS2812FX effects only use up to 3 different colors per segment. If you want more colors you have two options
1) divide your string of LEDs into segments and assign different colors to each segment
2) create custom effects to utilize as many colors as you like

Here is an example of a custom theater chase effect, which utilizes 5 colors. That might get you started.
```c++
#include <WS2812FX.h>

#define LED_COUNT 16
#define LED_PIN 22

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(32);

  ws2812fx.setCustomMode(myChase);
  ws2812fx.setSegment(0, 0, LED_COUNT - 1, FX_MODE_CUSTOM, RED, 200);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

uint32_t colors[] = { 0x00c3cc, 0x0e131f, 0x3372e3, 0xff0035, 0x7426ef };

uint16_t myChase() {
  WS2812FX::Segment* seg = ws2812fx.getSegment();
  WS2812FX::Segment_runtime* segrt = ws2812fx.getSegmentRuntime();
  int seglen = seg->stop - seg->start + 1;
  int numColors = sizeof(colors) / sizeof(colors[0]);

  for(uint16_t i=0; i < seglen; i++) {
    int colorIndex = (segrt->counter_mode_call + i) % numColors;
    ws2812fx.setPixelColor(seg->stop - i, colors[colorIndex]);
  }

  return seg->speed;
}
```

TheYigidOS, 2024-08-07:
Thanks a lot  it works fine!

TheYigidOS, 2024-08-10:
what if i want to set 2 colors to any kind of preset FX?

TheYigidOS, 2024-08-17:
sorry i have problem on my internet connection. i mean u write a lot of preset fx, how can i set color pallets for that like wled?

moose4lord, 2024-08-17:
WLED's borrowed WS2812FX's effects (which use the three color scheme), and also borrowed FASTled effects (which use FASTled's more robust color palettes). The two palette implementations are not interchangeable.

WS2812FX effects are limited to a maximum of three colors per segment (most effects only make use of two). I suppose you could create a palette of three colors and use it in your setSegment() functions. Like this maybe:
```c++
#include <WS2812FX.h>

#define LED_PIN   22  // digital pin used to drive the LED strip
#define LED_COUNT 30  // number of LEDs on the strip
#define DATA_PIN_MOSFET 23

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

// palettes
const uint32_t palettes[][3] = {
  { RED, GREEN, BLUE },             // PRIMARY
  { YELLOW, CYAN, MAGENTA },        // SECONDARY
  { PURPLE, ORANGE, PINK },         // WEIRD
  { 0xf08000, 0x448211, 0x900820 }  // CRAZY
};

// palette names used index into the palettes[] array for convenience
#define PRIMARY   0
#define SECONDARY 1
#define WEIRD     2
#define CRAZY     3

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(32);

  // parameters: index, start, stop, mode, color, speed
  ws2812fx.setSegment(0,  0,  9, FX_MODE_BLINK, palettes[0],         1000);
  ws2812fx.setSegment(1, 10, 19, FX_MODE_SCAN,  palettes[SECONDARY], 1000);
  ws2812fx.setSegment(2, 20, 29, FX_MODE_COMET, palettes[WEIRD],     1000);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```

If you want to have more control over palettes (and bigger palettes), you might want to look into the FASTled lib.

TheYigidOS, 2024-08-18:
Grazzie mille, thank you for your time and interest you make Great job...

closed 2024-08-18 by TheYigidOS