kitesurfer1404/WS2812FX #360: No longer compiling with VS code / PIO

snowrodeo, 2024-10-18:
I did an update and now the latest lib no longer compiles.  Here are the errors I am seeing
```
/Users/erikolson/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:52:3: error: expected unqualified-id before '(' token
   ({ __typeof__ (a) _a = (a); \
   ^
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:52:6: error: expected primary-expression before '__typeof__'
   ({ __typeof__ (a) _a = (a); \
      ^~~~~~~~~~
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:52:6: error: expected '}' before '__typeof__'
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:52:6: error: expected ')' before '__typeof__'
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:54:5: error: '_a' was not declared in this scope
     _a > _b ? _a : _b; })
     ^~
.pio/libdeps/ESPChecker_v4/WS2812FX/src/WS2812FX.h:54:5: note: suggested alternative: '_b'
```

snowrodeo, 2024-10-22:
I just created a PR for a fix to this issue.  Not sure how elegant my fix is, but it does the trick for me.

snowrodeo, 2024-10-22:
The compilers/versions must be different between Arduino IDE and Platformio.  I will dig in on that a bit and see if I can get the error to happen in IDE OR make a simple case in Platformio.  In the meantime, I will change my solve to use std::max and check it.  That would be a better solution compared to my hack.

snowrodeo, 2024-10-22:
New PR with std::max usage implemented

moose4lord, 2024-10-24:
OK, the switch to std::min/std::max doesn't work for AVR boards, since AVR boards don't seem to support the C++ STL. So I'll have to rethink that.

I'm back to trying to recreate the issue with PlatformIO, which I normally don't use. I installed PlatformIO and created a project with a simple WS2812FX sketch. It builds, uploads and runs on my ESP32 with no issues. I'm not sure what I'm doing "right".

PlatformIO prints this info when I compile the sketch:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/mhetesp32minikit.html
PLATFORM: Espressif 32 (3.5.0) > MH ET LIVE ESP32MiniKit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
 ```
The Arduino_ESP core v1.0.6 and ESP IDF v3.5.0 are quite old. Should I try upgrading to new versions? How would I do that?

Any help with reproducing the issue on my end would be appreciated.

snowrodeo, 2024-10-25:
I will work on the platformio stuff next Monday when work schedule frees up.

Do you want to try the first PR on the AVR platform?  This is a new macro that works on the ESP32.  It might work on the AVR as well.  Since we can’t use STD on AVR, this might be the path