kitesurfer1404/WS2812FX #366: Cannot compile for Arduino MEGA 2560

wbkwbk, 2024-11-19:
First of all the library is awesome. Thank to all contributors - great work!
I've used the library for several projects. 
For my latest I have to use the  Arduino MEGA 2560. Unfortunately the library does not compile.
It would be awesome if this issue could be fixed. 
Many thanks
In file included from C:\data\programming\LOTRGIT\Production\EffectsController\WS2812Wrapper.h:76:0,
                 from C:\data\programming\LOTRGIT\Production\EffectsController\EffectsController.ino:1:
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:143:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 0},  // K0 62
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:144:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 1},  // K1 63
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:145:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 2},  // K2 64
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:146:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 3},  // K3 65
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:147:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 4},  // K4 66
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:148:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 5},  // K5 67
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:149:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 6},  // K6 68
            ^~~~
C:\Users\wbkwb\Documents\Arduino\libraries\WS2812FX\src/WS2812FX.h:95:30: error: lvalue required as unary '&' operand
 #define PINK       (uint32_t)0xFF1493
                              ^
C:\Users\wbkwb\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:150:12: note: in expansion of macro 'PINK'
   {&DDRK, &PINK, &PORTK, 7}   // K7 69
            ^~~~

moose4lord, 2024-11-19:
This is an unfortunate name collision between the WS2812FX library and the SD library. Both libraries use the name "PINK" for different purposes. WS2812FX defines it as a color, while SD uses it to work with the Port K Input Pins Address of the Mega 2560.

The easiest fix for this is to rearrange the #include files in you sketch so that WS2812fx.h comes after SD.h.
```c++
#include <SD.h>
#include <WS2812FX.h>
```

wbkwbk, 2024-11-19:
Hi Keith
GREAT! Many thanks for your fast reply - you made my day!
Many thanks again for this great library and also to all the other contributors.
Best Regards
Martin

closed 2024-11-19 by wbkwbk