kitesurfer1404/WS2812FX #369: increaseSpeed and other options

ubear7, 2025-02-15:
I have read everything and searched everywhere. I am obviously a novice. Can someone supply an example using the increase_Speed option. Also maybe include one that will cycle a specific amount of time then stop. I see many options but do not understand how to implement them as I get an error when compiling.

moose4lord, 2025-02-15:
Here is an example sketch that changes the animation speed.  It uses a timer to update the speed parameter every 1 second, first increasing the speed parameter until it reaches 8000, then decreasing the speed parameter until it reaches 200.

It also has a snippet of code to stop the animation after 2 minutes.

```c++
#include <WS2812FX.h>

#define LED_PIN   14  // digital pin used to drive the LED strip
#define LED_COUNT 64  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

unsigned long timer = 0;  // timer used to change speed
bool isSpeedIncreasing = true;  //  flag to track if the speed is increasing or decreasing

void setup() {
  Serial.begin(115200);

  ws2812fx.init();
  ws2812fx.setBrightness(4);

  ws2812fx.setSegment(0,  0,  LED_COUNT-1, FX_MODE_SCAN, BLUE, 1000);

  ws2812fx.start();
}

void loop() {
  if(!ws2812fx.isRunning()) return;  // if the LED are NOT running, just return

  unsigned long now = millis();  // get number of milliseconds since power up

  // if powered up for more then 120000ms (2 minutes), than stop
  if(now > 120000) {
    ws2812fx.stop();
    Serial.println("The LEDs are stopped");
  } else {  // otherwise run the LED animation
    ws2812fx.service();

    if(now > timer + 1000) {         // change speed every 1000ms
      if(isSpeedIncreasing) {
        ws2812fx.increaseSpeed(200); // increase speed (LED animation gets SLOWER)
      } else {
        ws2812fx.decreaseSpeed(200); // decrease speed (LED animation gets FASTER)
      }

      // if speed is greater than 8000, switch to decreasing speed
      if(ws2812fx.getSpeed() >= 8000) isSpeedIncreasing = false;
      // if speed is less than 200, switch to increasing speed
      if(ws2812fx.getSpeed() <=  200) isSpeedIncreasing = true;
  
      // print the update speed parameter
      Serial.print("Speed is ");
      Serial.println(ws2812fx.getSpeed());
    
      timer = now;  // reset the timer
      }
  }
}

```
Hope that helps.

ubear7, 2025-02-15:
Thank you very much for your quick and accurate help. It helped me a lot.
I'm sure if you post this as an example it would help many other beginners.